yubox-node-org/ESPAsyncWebServer #5: Always restarts on page refresh with the same exception. Even with new sketch. ( _sockIsReadable() )

zekageri, 2022-02-24:
Hi! If i refresh a page more then twice which includes three minified and gzipped files ( max 5kb each ! ) my esp always restarts with always the same exception:

```C
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d7579  PS      : 0x00060c30  A0      : 0x800e89b0  A1 
     : 0x3ffe8100
A2      : 0x00000000  A3      : 0x3ffc9b3c  A4      : 0x3ffe36e4  A5 
     : 0x025b69d9
A6      : 0x0000000d  A7      : 0x00000000  A8      : 0x800d7576  A9 
     : 0x3ffe80e0
A10     : 0x3ffe8134  A11     : 0x3f400c5a  A12     : 0x025b69d9  A13     : 0x3ffc9b3c
A14     : 0x00000000  A15     : 0x3ffe814c  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffa

ELF file SHA256: 0000000000000000

Backtrace: 0x400d7579:0x3ffe8100 0x400e89ad:0x3ffe8180 0x400e5f0d:0x3ffe81d0 0x400e5fd9:0x3ffe8210 0x400e6241:0x3ffe8250 0x4016ae62:0x3ffe8270 0x4016a9b2:0x3ffe82a0 0x4008a32e:0x3ffe82f0
  #0  0x400d7579:0x3ffe8100 in std::_Function_handler<void (AsyncWebServerRequest*), sSystem::adminEndpoints()::{lambda(AsyncWebServerRequest*)#3}>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) at lib\ArduinoJson-6.x\src/ArduinoJson/Numbers/parseNumber.hpp:144  
      (inlined by) _M_invoke at c:\users\pc\.platformio\packages\toolchain-xtensa32@2.50200.97\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #1  0x400e89ad:0x3ffe8180 in std::function<void (AsyncWebServerRequest*)>::operator()(AsyncWebServerRequest*) const at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebServer.cpp:197
      (inlined by) AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebHandlerImpl.h:132
  #2  0x400e5f0d:0x3ffe81d0 in AsyncWebServerRequest::_parseLine() at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebRequest.cpp:814
  #3  0x400e5fd9:0x3ffe8210 in AsyncWebServerRequest::_onData(void*, 
unsigned int) at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebRequest.cpp:814
  #4  0x400e6241:0x3ffe8250 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#6}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebRequest.cpp:814   
      (inlined by) _M_invoke at c:\users\pc\.platformio\packages\toolchain-xtensa32@2.50200.97\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #5  0x4016ae62:0x3ffe8270 in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at lib\AsyncTCPSock\src/AsyncTCP.cpp:360
      (inlined by) AsyncClient::_sockIsReadable() at lib\AsyncTCPSock\src/AsyncTCP.cpp:885
  #6  0x4016a9b2:0x3ffe82a0 in _asynctcpsock_task(void*) at lib\AsyncTCPSock\src/AsyncTCP.cpp:360
  #7  0x4008a32e:0x3ffe82f0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

For example this is the endpoint handler for a particular page:

```C
// hsh_Server.adminIsIn is just a flag.
// hsh_Server.adminLastExtendTime is a long

server.on("/admin", HTTP_GET, [](AsyncWebServerRequest *request){
    if( !request->authenticate("admin", "admin") ){
        return request->requestAuthentication();
    }
    if( hsh_Server.adminIsIn && millis() - hsh_Server.adminLastExtendTime < 2000 ){
        request->send(200, "text/html",
            "<head><meta content='width=device-width; initial-scale=1.0; maximum-scale=1.0; user-scalable=0;' name='viewport' />\
            <meta name='viewport' content='width=device-width'/></head>\
            <style>body{margin: 0px;padding: 0px;width: 100%;}\
                html {-webkit-box-sizing: border-box;-moz-box-sizing: border-box;box-sizing: border-box;}\
            </style>\
            <div style='width:100%;height:100%;display:flex;justify-content:center;align-items:center;font-weight:bold;'>\
                <h1 style='text-align: center;'>An admin is already logged in.</h1>\
            </div>");
    }else{
        String path = hsh_fileSystem.correctPath("/admin/bundles/zipped/admin.html");
        AsyncWebServerResponse *response = request->beginResponse(LITTLEFS, path, getMimeType(path));
        response->addHeader("Server","HsH Server");
        response->addHeader("Content-Encoding","gzip");
        request->send(response);
    }
});
```

With two files which minified and gzipped. ( max 5kb each )

```C
server.on("/adminStyle.bundle.css", HTTP_GET, [](AsyncWebServerRequest *request){
    String path = hsh_fileSystem.correctPath("/admin/bundles/zipped/adminStyle.bundle.css");
    AsyncWebServerResponse *response = request->beginResponse(LITTLEFS, path, getMimeType(path));
    response->addHeader("Server","HsH Server");
    response->addHeader("Content-Encoding","gzip");
    request->send(response);
});

server.on("/adminScript.bundle.js", HTTP_GET, [](AsyncWebServerRequest *request){
    String path = hsh_fileSystem.correctPath("/admin/bundles/zipped/adminScript.bundle.js");
    AsyncWebServerResponse *response = request->beginResponse(LITTLEFS, path, getMimeType(path));
    response->addHeader("Server","HsH Server");
    response->addHeader("Content-Encoding","gzip");
    request->send(response);
});
```

There is nothing more to it except the Websocket connection which will happen almost immidiately after page load.
But the crash is not on the websocket open callback but right on the endpoint handler.

**Only two or three refresh what it needs to reboot the esp with this exception code.**

### Can you help?

Pablo2048, 2022-02-25:
According to the crash backtrace there is ArduinoJSON involved, but I don't see any use of ArduinoJSON in your handlers. So what is `sSystem::adminEndpoints()` and why is he requesting some number via ArduinoJSON?

Pablo2048, 2022-02-25:
... so I dig deeper in Your repositories, an I think that the `sSystem::adminEndpoint()` came from https://github.com/zekageri/Mood_Lamp/blob/cad1e607ed872de597424a35c419f228ab4bee04/src/utilities/serverSystem.cpp#L42 , but so far no ArduinoJSON call...

zekageri, 2022-02-25:
Yeah, arduinoJSON call is in the websocket handler. All my websocket messages are JSON strings. But the websocket callback never executed because the crash happens right inside the adminEndpoint handler. Before the page gets trought.

Here is the socket callback.

```C
void hsh_adminSocketEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len){
    if(type == WS_EVT_CONNECT){
        #if SS_DEBUG_MODE
            Serial.println("WebSocket admin connected");
            hsh_Server.adminIsIn = true;
            hsh_Server.adminLastExtendTime = millis();
        #endif
    }else if(type == WS_EVT_DISCONNECT){
        #if SS_DEBUG_MODE
            Serial.println("WebSocket admin disconnected");
            hsh_Server.adminIsIn = false;
        #endif
    }else if(type == WS_EVT_DATA){
        AwsFrameInfo * info = (AwsFrameInfo*)arg;
        for (size_t i = 0; i < len; i++){
            hsh_Server.adminSocketBuffer[hsh_Server.adminSocketBuffIndex] = data[i];
            hsh_Server.adminSocketBuffIndex++;
        }
        if( hsh_Server.adminSocketBuffIndex >= info->len ){
            hsh_Server.adminSocketBuffer[hsh_Server.adminSocketBuffIndex] = '\0';
            hsh_Server.adminSocketBuffIndex = 0;

            DynamicJsonDocument socketJsonBuffer(ADMIN_SOCKET_INCOMING_BUFFER_SIZE);
            DeserializationError error = deserializeJson(socketJsonBuffer, hsh_Server.adminSocketBuffer);
            if(error){
                #if SS_DEBUG_MODE
                    Serial.println("Failed to parse JSON");
                #endif
            }else{
                JsonObject socketMessage = socketJsonBuffer.as<JsonObject>();
                String type = socketMessage["type"].as<String>();
                if( type == "getDirectory" ){
                    hsh_Server.socketSendString(hsh_fileSystem.listDir(socketMessage["path"].as<const char*>()).c_str(), 1, client->id());
                }else if( type == "extendAdmin" ){
                    hsh_Server.adminLastExtendTime = millis();
                }else if( type == "getFile" ){
                    hsh_Server.socketSendString(hsh_fileSystem.getFile(socketMessage["filePath"].as<const char*>()).c_str(), 1, client->id());
                }else if( type == "downloadFile" ){
                    hsh_Server.socketSendString(hsh_fileSystem.downloadFile(socketMessage["filePath"].as<const char*>()).c_str(), 1, client->id());
                }else if( type == "massDelete" ){
                    DynamicJsonDocument massDelBuffer(500);
                    JsonArray files = socketMessage["files"].as<JsonArray>();
                    JsonArray filesDeleted = massDelBuffer.createNestedArray("filesDeleted");
                    int successCount = 0;
                    for(int i = 0; i < files.size(); i++){
                        if( hsh_fileSystem.deleteFile(files[i].as<const char*>()) ){
                            successCount++;
                            filesDeleted.add(files[i]);
                        }
                    }
                    massDelBuffer["type"] = "massDelete";
                    massDelBuffer["successCount"]   = successCount;
                    massDelBuffer["totalCount"]     = files.size();
                    serializeJson(massDelBuffer, hsh_Server.adminSocketBuffer);
                    hsh_Server.socketSendString(hsh_Server.adminSocketBuffer, 1, client->id());
                }else if( type == "saveConfig" ){

                }else if( type == "logOut" ){
                    hsh_Server.adminIsIn = false;
                    hsh_Server.adminLastExtendTime = 0;
                    DynamicJsonDocument logOutResp(500);
                    logOutResp["type"]      = "logOut";
                    logOutResp["success"]   = true;
                    serializeJson(logOutResp, hsh_Server.adminSocketBuffer);
                    hsh_Server.socketSendString(hsh_Server.adminSocketBuffer, 1, client->id());
                }else if( type == "managerCommand" ){
                    String command = socketMessage["command"].as<String>();
                    Serial.println(command);
                    if( command == "hwReset" ){
                        hsh_Performance.beginHwReset();
                    }else if( command == "restart" ){
                        hsh_Performance.beginRestart();
                    }
                }
            }
        }
    }
}
```

**There is extensive use of arduinoJSON inside the socket handler but as i said, the program never gets here. The webpage even delays the socket connection after page load.**

zekageri, 2022-02-25:
What you have seen is the main branch. The current adminEndpoints looks like what i copied in here before:

```C
void sSystem::adminEndpoints(){
    server.on("/admin", HTTP_GET, [](AsyncWebServerRequest *request){
        if( !request->authenticate("admin", "admin") ){
            return request->requestAuthentication();
        }
        if( hsh_Server.adminIsIn && millis() - hsh_Server.adminLastExtendTime < 2000 ){
            request->send(200, "text/html",
                "<head><meta content='width=device-width; initial-scale=1.0; maximum-scale=1.0; user-scalable=0;' name='viewport' />\
                <meta name='viewport' content='width=device-width'/></head>\
                <style>body{margin: 0px;padding: 0px;width: 100%;}\
                    html {-webkit-box-sizing: border-box;-moz-box-sizing: border-box;box-sizing: border-box;}\
                </style>\
                <div style='width:100%;height:100%;display:flex;justify-content:center;align-items:center;font-weight:bold;'>\
                    <h1 style='text-align: center;'>An admin is already logged in.</h1>\
                </div>");
        }else{
            String path = hsh_fileSystem.correctPath("/admin/bundles/zipped/admin.html");
            AsyncWebServerResponse *response = request->beginResponse(LITTLEFS, path, getMimeType(path));
            response->addHeader("Server","HsH Server");
            response->addHeader("Content-Encoding","gzip");
            request->send(response);
        }
    });

    server.on("/adminStyle.bundle.css", HTTP_GET, [](AsyncWebServerRequest *request){
        String path = hsh_fileSystem.correctPath("/admin/bundles/zipped/adminStyle.bundle.css");
        AsyncWebServerResponse *response = request->beginResponse(LITTLEFS, path, getMimeType(path));
        response->addHeader("Server","HsH Server");
        response->addHeader("Content-Encoding","gzip");
        request->send(response);
    });

    server.on("/adminScript.bundle.js", HTTP_GET, [](AsyncWebServerRequest *request){
        String path = hsh_fileSystem.correctPath("/admin/bundles/zipped/adminScript.bundle.js");
        AsyncWebServerResponse *response = request->beginResponse(LITTLEFS, path, getMimeType(path));
        response->addHeader("Server","HsH Server");
        response->addHeader("Content-Encoding","gzip");
        request->send(response);
    });
}
```

Pablo2048, 2022-02-25:
According to the crash stacktrace the ArduinoJSON was called so my suggestion is: rewrite the code to not to use lambdas for `server.on()` calls - it's more debug-friendly not to see `lambda()#x` in the backtrace and post the new exception dump.

zekageri, 2022-02-25:
Oh. I did not know I could write the endpoints without lambda. I will investigate it

zekageri, 2022-02-25:
yeah. with callbacks... silly me

```C
void sSystem::adminPageHandler(AsyncWebServerRequest *request){
    if( !request->authenticate("admin", "admin") ){
        return request->requestAuthentication();
    }
    if( hsh_Server.adminIsIn && millis() - hsh_Server.adminLastExtendTime < 2000 ){
        request->send(200, "text/html",
            "<head><meta content='width=device-width; initial-scale=1.0; maximum-scale=1.0; user-scalable=0;' name='viewport' />\
            <meta name='viewport' content='width=device-width'/></head>\
            <style>body{margin: 0px;padding: 0px;width: 100%;}\
                html {-webkit-box-sizing: border-box;-moz-box-sizing: border-box;box-sizing: border-box;}\
            </style>\
            <div style='width:100%;height:100%;display:flex;justify-content:center;align-items:center;font-weight:bold;'>\
                <h1 style='text-align: center;'>An admin is already logged in.</h1>\
            </div>");
    }else{
        String path = hsh_fileSystem.correctPath("/admin/bundles/zipped/admin.html");
        AsyncWebServerResponse *response = request->beginResponse(LITTLEFS, path, getMimeType(path));
        response->addHeader("Server","HsH Server");
        response->addHeader("Content-Encoding","gzip");
        request->send(response);
    }
}

void sSystem::adminPageScriptHandler(AsyncWebServerRequest *request){
    String path = hsh_fileSystem.correctPath("/admin/bundles/zipped/adminStyle.bundle.css");
    AsyncWebServerResponse *response = request->beginResponse(LITTLEFS, path, getMimeType(path));
    response->addHeader("Server","HsH Server");
    response->addHeader("Content-Encoding","gzip");
    request->send(response);
}

void sSystem::adminPageStyleHandler(AsyncWebServerRequest *request){
    String path = hsh_fileSystem.correctPath("/admin/bundles/zipped/adminScript.bundle.js");
    AsyncWebServerResponse *response = request->beginResponse(LITTLEFS, path, getMimeType(path));
    response->addHeader("Server","HsH Server");
    response->addHeader("Content-Encoding","gzip");
    request->send(response);
}

void sSystem::adminEndpoints(){
    server.on("/admin", HTTP_GET, adminPageHandler);
    server.on("/adminStyle.bundle.css", HTTP_GET, adminPageScriptHandler);
    server.on("/adminScript.bundle.js", HTTP_GET, adminPageStyleHandler);
}
```

### crash log: 

```C
lib\LITTLEFS\src\lfs.c:1076:error: Corrupted dir pair at {0x0, 0x1}
lib\LITTLEFS\src\lfs.c:1076:error: Corrupted dir pair at {0x0, 0x1}
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.   
Core 0 register dump:
PC      : 0x400d750d  PS      : 0x00060e30  A0      : 0x8018d13a  A1      : 0x3ffe80e0
A2      : 0x3ffbc264  A3      : 0x00000000  A4      : 0x3ffc9b3c  A5      : 0x00000001
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d750a  A9      : 0x3ffe80c0
A10     : 0x3ffe80f0  A11     : 0x3f400c49  A12     : 0x85438723  A13     : 0x3ffc9b3c
A14     : 0x00000000  A15     : 0x3ffe812c  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffa

ELF file SHA256: 0000000000000000

Backtrace: 0x400d750d:0x3ffe80e0 0x4018d137:0x3ffe8160 0x400e8add:0x3ffe8180 0x400e603d:0x3ffe81d0 0x400e6109:0x3ffe8210 0x400e6371:0x3ffe8250 0x4016af4e:0x3ffe8270 0x4016aa9e:0x3ffe82a0 0x4008a32e:0x3ffe82f0
  #0  0x400d750d:0x3ffe80e0 in sSystem::adminPageHandler(AsyncWebServerRequest*) at lib\ArduinoJson-6.x\src/ArduinoJson/Numbers/parseNumber.hpp:144
  #1  0x4018d137:0x3ffe8160 in std::_Function_handler<void (AsyncWebServerRequest*), 
void (*)(AsyncWebServerRequest*)>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) at lib\ArduinoJson-6.x\src/ArduinoJson/Numbers/parseNumber.hpp:144
  #2  0x400e8add:0x3ffe8180 in std::function<void (AsyncWebServerRequest*)>::operator()(AsyncWebServerRequest*) const at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebServer.cpp:197
      (inlined by) AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at 
lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebHandlerImpl.h:132       
  #3  0x400e603d:0x3ffe81d0 in AsyncWebServerRequest::_parseLine() at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebRequest.cpp:814
  #4  0x400e6109:0x3ffe8210 in AsyncWebServerRequest::_onData(void*, unsigned int) at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebRequest.cpp:814        
  #5  0x400e6371:0x3ffe8250 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#6}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebRequest.cpp:814
      (inlined by) _M_invoke at c:\users\pc\.platformio\packages\toolchain-xtensa32@2.50200.97\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #6  0x4016af4e:0x3ffe8270 in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at lib\AsyncTCPSock\src/AsyncTCP.cpp:360
      (inlined by) AsyncClient::_sockIsReadable() at lib\AsyncTCPSock\src/AsyncTCP.cpp:885
  #7  0x4016aa9e:0x3ffe82a0 in _asynctcpsock_task(void*) at lib\AsyncTCPSock\src/AsyncTCP.cpp:360
  #8  0x4008a32e:0x3ffe82f0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

zekageri, 2022-02-25:
It still complaining about the adminPageHandler callback function which still not using arduinoJSON

```C
/* 
Functions used by adminPageHandler():

String fSystem::correctPath(const char* path){
    if( LITTLEFS.exists(path) ){
        return String(path);
    }
    return "/notFound/notFound.html";
}

String getMimeType(String path){
    if(path.endsWith(".html")) return "text/html";
    else if(path.endsWith(".css")) return "text/css";
    else if(path.endsWith(".js")) return "application/javascript";
    else if(path.endsWith(".ico")) return "image/x-icon";
    else if(path.endsWith(".gz")) return "application/x-gzip";
    return "text/plain";
}

No arduinoJSON
*/
void sSystem::adminPageHandler(AsyncWebServerRequest *request){
    if( !request->authenticate("admin", "admin") ){
        return request->requestAuthentication();
    }
    if( hsh_Server.adminIsIn && millis() - hsh_Server.adminLastExtendTime < 2000 ){
        request->send(200, "text/html",
            "<head><meta content='width=device-width; initial-scale=1.0; maximum-scale=1.0; user-scalable=0;' name='viewport' />\
            <meta name='viewport' content='width=device-width'/></head>\
            <style>body{margin: 0px;padding: 0px;width: 100%;}\
                html {-webkit-box-sizing: border-box;-moz-box-sizing: border-box;box-sizing: border-box;}\
            </style>\
            <div style='width:100%;height:100%;display:flex;justify-content:center;align-items:center;font-weight:bold;'>\
                <h1 style='text-align: center;'>An admin is already logged in.</h1>\
            </div>");
    }else{
        String path = hsh_fileSystem.correctPath("/admin/bundles/zipped/admin.html");
        AsyncWebServerResponse *response = request->beginResponse(LITTLEFS, path, getMimeType(path));
        response->addHeader("Server","HsH Server");
        response->addHeader("Content-Encoding","gzip");
        request->send(response);
    }
}
```

zekageri, 2022-02-25:
Upgraded ArduinoJSON library to the latest, now here is the crash report:

```C
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was 
unhandled.
Core 0 register dump:
PC      : 0x400d7691  PS      : 0x00060830  A0      : 0x8018d116  A1    
  : 0x3ffe80e0
A2      : 0x3ffec57c  A3      : 0x00000000  A4      : 0x3ffc9b3c  A5    
  : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d768e  A9    
  : 0x3ffe80c0
A10     : 0x3ffe80f0  A11     : 0x3f400c49  A12     : 0x64cd35cc  A13   
  : 0x3ffc9b3c
A14     : 0x00000000  A15     : 0x3ffe812c  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x400d7691:0x3ffe80e0 0x4018d113:0x3ffe8160 0x400e8ad9:0x3ffe8180 0x400e6039:0x3ffe81d0 0x400e6105:0x3ffe8210 0x400e636d:0x3ffe8250 0x4016af4a:0x3ffe8270 0x4016aa9a:0x3ffe82a0 0x4008a32e:0x3ffe82f0        
  #0  0x400d7691:0x3ffe80e0 in sSystem::adminPageScriptHandler(AsyncWebServerRequest*) at lib\ArduinoJson-6.x\src/ArduinoJson/Serialization/Writers/ArduinoStringWriter.hpp:30
  #1  0x4018d113:0x3ffe8160 in std::_Function_handler<void (AsyncWebServerRequest*), void (*)(AsyncWebServerRequest*)>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) at lib\ArduinoJson-6.x\src/ArduinoJson/Serialization/Writers/ArduinoStringWriter.hpp:30
  #2  0x400e8ad9:0x3ffe8180 in std::function<void (AsyncWebServerRequest*)>::operator()(AsyncWebServerRequest*) const at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebServer.cpp:197
      (inlined by) AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebHandlerImpl.h:132
  #3  0x400e6039:0x3ffe81d0 in AsyncWebServerRequest::_parseLine() at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebRequest.cpp:814
  #4  0x400e6105:0x3ffe8210 in AsyncWebServerRequest::_onData(void*, unsigned int) at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebRequest.cpp:814
  #5  0x400e636d:0x3ffe8250 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, 
unsigned int)#6}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at lib\ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup\src/WebRequest.cpp:814
      (inlined by) _M_invoke at c:\users\pc\.platformio\packages\toolchain-xtensa32@2.50200.97\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #6  0x4016af4a:0x3ffe8270 in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at lib\AsyncTCPSock\src/AsyncTCP.cpp:360
      (inlined by) AsyncClient::_sockIsReadable() at lib\AsyncTCPSock\src/AsyncTCP.cpp:885
  #7  0x4016aa9a:0x3ffe82a0 in _asynctcpsock_task(void*) at lib\AsyncTCPSock\src/AsyncTCP.cpp:360
  #8  0x4008a32e:0x3ffe82f0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

zekageri, 2022-02-25:
It looks the same but in another callback

Pablo2048, 2022-02-25:
I thought that it may be problem in `->authenticate()` or `->requestAuthentication()`, but after your ArduinoJSON library update I'm out of ideas. Did you try to check the line numbers in the files given by the backtrace if something doesn't pop to you?

zekageri, 2022-02-25:
Upgraded arduinojson lib and defined to use long long.

```C
#define ARDUINOJSON_USE_LONG_LONG 1
#include <ArduinoJson.h>
```

The problem was delayed but it occoured after like 5-6 page request.

The line numbers dont tell me anything sadly. I thought that it could be because the page requesting a config file which using epoch times and it would give an explanation but after i defined it to use long long i dont know.

The fact that the crash says that its happening inside the adminPageHandler is really annoying. But the program really does not run further then that

zekageri, 2022-02-25:
For some reason LITTLEFS has no chill and on every page refresh i get corrupted dir pairs.....

```C
WebSocket admin connected
lib\LITTLEFS\src\lfs.c:1076:error: Corrupted dir pair at {0x1, 0x0}
lib\LITTLEFS\src\lfs.c:1076:error: Corrupted dir pair at {0x1, 0x0}
lib\LITTLEFS\src\lfs.c:1076:error: Corrupted dir pair at {0x1, 0x0}
WebSocket admin connected
WebSocket admin connected
lib\LITTLEFS\src\lfs.c:1076:error: Corrupted dir pair at {0x1, 0x0}
lib\LITTLEFS\src\lfs.c:1076:error: Corrupted dir pair at {0x1, 0x0}
WebSocket admin connected
lib\LITTLEFS\src\lfs.c:1076:error: Corrupted dir pair at {0x1, 0x0}
lib\LITTLEFS\src\lfs.c:1076:error: Corrupted dir pair at {0x1, 0x0}
lib\LITTLEFS\src\lfs.c:1076:error: Corrupted dir pair at {0x1, 0x0}
WebSocket admin connected
```

zekageri, 2022-02-25:
Maybe the problem is that when i refresh fairly rapidly, the last socket send try to send the message to the client before but that client is disconnected and thats why the arduinojson error message? idk :'D

zekageri, 2022-02-25:
With this function on loop:

```C
void sSystem::cleanUnusedSocketConnections(){
    hsh_Socket.cleanupClients();
    hsh_adminSocket.cleanupClients();
}
```

And with a check before textAll:

```C
void sSystem::socketSendString(const char* message,int socket,int clientID){
    socketLock();
    if(socket == ADMIN_SOCKET && hsh_adminSocket.count() > 0){
        if(clientID == -1){
            hsh_adminSocket.textAll(message);
        }else{
            hsh_adminSocket.text(clientID,message);
        }
    }
    socketUnlock();
}
```

It seems more stable??

Pablo2048, 2022-02-25:
Unfortunately it seems like I'm unable to help you more... :-( So my last idea - check the filesystem and/or reformat it. Maybe it is the source of the crash...

zekageri, 2022-02-25:
Thank you for all of that. I did checked the file system, reformatted it.
I even built a class around LITTLEFS in order to protect the calls with a semaphore.
It does not prevent the ESPAsyncWebServer calls tho. :(

zekageri, 2022-02-25:
Nice. there are some other errors too.

```C
ELF file SHA256: 0000000000000000

Backtrace: 0x400890a0:0x3ffe8140 0x4008931d:0x3ffe8160 0x4008a059:0x3ffe8180 0x4016ae4c:0x3ffe81c0 0x4016af96:0x3ffe81e0 0x4016b031:0x3ffe8210 0x4016ba01:0x3ffe8240 0x4016ab45:0x3ffe82a0 0x4008a32e:0x3ffe82f0
  #0  0x400890a0:0x3ffe8140 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x4008931d:0x3ffe8160 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x4008a059:0x3ffe8180 in xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2038
  #3  0x4016ae4c:0x3ffe81c0 in AsyncClient::_clearWriteQueue() at lib\AsyncTCPSock\src/AsyncTCP.cpp:360
  #4  0x4016af96:0x3ffe81e0 in AsyncClient::_error(signed char) at lib\AsyncTCPSock\src/AsyncTCP.cpp:360
  #5  0x4016b031:0x3ffe8210 in AsyncClient::_notifyWrittenBuffers(std::deque<AsyncClient::notify_writebuf, std::allocator<AsyncClient::notify_writebuf> >&, int) at lib\AsyncTCPSock\src/AsyncTCP.cpp:360   
  #6  0x4016ba01:0x3ffe8240 in AsyncClient::_sockIsWriteable() at lib\AsyncTCPSock\src/AsyncTCP.cpp:360
  #7  0x4016ab45:0x3ffe82a0 in _asynctcpsock_task(void*) at lib\AsyncTCPSock\src/AsyncTCP.cpp:360     
  #8  0x4008a32e:0x3ffe82f0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

zekageri, 2022-03-02:
It seems to me that if i request LITTLEFS for it's `usedBytes()` to frequently, it is causing the issues.

`LITTLEFS.usedBytes();`

I'am testing now. Before i was pulled this every second, now iam pulling it every 15k ms. Seems fine so far. No crash but i'm not sure yet.

zekageri, 2022-04-29:
The issue was because LITTLEFS doesn't use semaphores to protect itself from multi access.
Upgraded to LittleFs and it seems to work properly. Another issue occoured hoverwer.

closed 2022-04-29 by zekageri